ragnraok/RxCamera #1: Support record video

longbkiter07, 2016-01-11:
Your library is pretty cool. It's very easy to use. But it only supports taking pictures. It's better if you have a plan to support recording video. Waiting for you :)

ragnraok, 2016-01-16:
That is a good point, but RxCamera may only use MediaRecorder to record the video, because record video in android is pretty complicated, and I don't think integrate the ffmpeg in RxCamera is a good idea :)

sibeliusseraphini, 2016-03-15:
@longbkiter07 you should take a look on http://developer.android.com/reference/android/media/MediaRecorder.html

If you wanna, send a PR for this!

Take a look at https://realm.io/news/droidcon-gomez-functional-reactive-programming/, so you can understand better how Rx works

sibeliusseraphini, 2016-03-16:
@longbkiter07 take a look of this

https://gist.github.com/adelnizamutdinov/d1a99de548ee9629513d (RxMediaRecorder)
https://gist.github.com/adelnizamutdinov/8bc276f2eda05536bf0c (RxMediaPlayer)