ragnraok/RxCamera #2: Support for camera2

bunjix, 2016-01-12:
Since API 21, [camera2](https://developer.android.com/reference/android/hardware/camera2/package-summary.html) replaces the [Camera](https://developer.android.com/reference/android/hardware/Camera.html) class.

ragnraok, 2016-01-16:
the camera2 API only support after API-21, and currently not all devices support all features in camera2, when use the developers should query the support level of camera2: http://source.android.com/devices/camera/versioning.html#camera_api2_capabilities_and_support_levels, which is pretty hard to use, so the RxCamera currently doesn't have the plan to support this

bunjix, 2016-01-16:
Ok, thanks. This lib works very well.

closed 2016-01-16 by ragnraok

sibeliusseraphini, 2016-03-08:
@ragnraok I think you should let this issue open, and maybe somebody could fine a clever way to provide the same RxCamera API using camera2 or camera1 depending on the API level