ragnraok/RxCamera #5: RxCamera.CameraInternal: open camera failed: Fail to connect to camera service

spirosoik, 2016-01-20:
I ran the example and the camera cannot open in a samsung lg tab g pad 7.0

ragnraok, 2016-01-23:
did you add ``<uses-permission android:name="android.permission.CAMERA" />`` in your AndroidManifest.xml?

spirosoik, 2016-01-23:
Hi @ragnraok I found the problem I think. My case is to open the camera when the screen is visible not with a button open. I understood that the problem is that surface is not available so I added the listener for ```TextureView```. When the listener says that I am available then I open the camera without a problem. Just I suggest to add this case in your README. 

By the way great job thanks for the great repo. You can close the issue

ragnraok, 2016-01-23:
RxCamera will add a internal surface callback if you set ``HandleSurfaceEvent`` to true, and will call ``startPreview`` when the surface is available, but the ``camera.open`` can be called at any time whether the surface is available, may be the internal logic of open camera has something wrong, I will check that, thanks!

closed 2016-01-23 by ragnraok

spirosoik, 2016-01-30:
@ragnraok Yes I know but something is not working properly I suppose.

sibeliusseraphini, 2016-03-08:
@spirosoik could you provide your solution?

sibeliusseraphini, 2016-03-08:
@ragnraok do you have an example without an open button?

samy-baili, 2016-12-13:
Anyone find a solution for this issue ?