ragnraok/RxCamera #11: Flash availability

EzequielAdrianM, 2016-03-09:
This is a question. Is flash available to turn on/off while taking photos?

sibeliusseraphini, 2016-03-14:
@EzequielAdrianM you should take a look at RxCameraActionBuilder

sibeliusseraphini, 2016-03-14:
```java
public Observable<RxCamera> flash() {
        return Observable.create(new Observable.OnSubscribe<RxCamera>() {

            @Override
            public void call(Subscriber<? super RxCamera> subscriber) {
                Camera.Parameters parameters = rxCamera.getNativeCamera().getParameters();

                parameters.setFlashMode(Camera.Parameters.FLASH_MODE_TORCH);
                
                rxCamera.getNativeCamera().setParameters(parameters);
                subscriber.onNext(rxCamera);
            }
        });
    }
```

You just need to add some checks and you are ready to go

closed 2016-03-15 by EzequielAdrianM

sibeliusseraphini, 2016-03-21:
@EzequielAdrianM the flash support was added on this commit https://github.com/ragnraok/RxCamera/blob/2a49ce63db417d8af0638f138bf3a2ca095457f1/lib/src/main/java/com/ragnarok/rxcamera/action/RxCameraActionBuilder.java