ragnraok/RxCamera #16: Suggest update RxJava and RxAndroid version.

wind0ws, 2016-03-21:
I see your lib use old version of rxjava and rxandroid.

And currently,I use your lib by this in gradle script:

```
compile 'io.reactivex:rxjava:1.1.1'
 compile 'io.reactivex:rxandroid:1.1.0'

    compile('com.ragnarok.rxcamera:lib:0.0.2'){
        exclude group: 'io.reactivex',module:'rxjava'
        exclude group: 'io.reactivex',module:'rxandroid'
    }
```

wind0ws, 2016-03-21:
By the way,thanks to ragnraok for your awesome RxCamera Lib.

sibeliusseraphini, 2016-03-21:
@wind0ws would you mind to send a PR for this?

wind0ws, 2016-03-21:
@sibeliusseraphini Sorry for my poor english.Does PR means Public-Relation?
  RxCamera depend on RxAndroid and RxJaAVA, RxAndroid depend on RxJava.

sibeliusseraphini, 2016-03-21:
PR - Pull Request

closed 2016-03-29 by wind0ws