ragnraok/RxCamera #19: periodicDataRequest where is the data?

EzequielAdrianM, 2016-03-30:
camera.request().periodicDataRequest(500).subscribe(new Action1<RxCameraData>() {
        @Override
        public void call(RxCameraData rxCameraData) {
            showLog("periodic request, cameradata.length: " + rxCameraData.cameraData.length);
                Bitmap bitmap = BitmapFactory.decodeByteArray(rxCameraData.cameraData, 0, rxCameraData.cameraData.length);

Gives this warning and then App Crashes:

    periodic request, cameradata.length: 460800
    --- SkImageDecoder::Factory returned null

Because rxCameraData.cameraData is NULL...

ragnraok, 2016-04-04:
the ``periodicDataRequest`` just return the raw camera capture data, which is in YUV420 format in most devices, so you can't directly decode it by ``BitmapFactory``

EzequielAdrianM, 2016-04-07:
Oh I see, and could you give me an example about converting YUV420 to Bitmap?

EzequielAdrianM, 2016-04-09:
I have tried with this but I don't know the width and height of Yuv420.

    Bitmap bm = Bitmap.createBitmap(width, height, Bitmap.Config.ARGB_8888);
    bm.copyPixelsFromBuffer(ByteBuffer.wrap(rawData));

ragnraok, 2016-04-09:
the width and height of the yuv420sp data just the preview size, and you can retrieve the transformation equation from [Google](https://www.google.co.jp/search?q=YUV420sp+to+rgb&oq=YUV420sp+to+rgb&gs_l=serp.3...1036.3047.0.3198.9.7.0.0.0.0.547.547.5-1.1.0....0...1c.1.64.serp..8.1.546.8LQnkJuJaDs)

closed 2016-04-12 by ragnraok

EzequielAdrianM, 2016-04-13:
SOLVED! Thankyou @ragnraok !

    public void call(RxCameraData rxCameraData) {
        try {
            Camera.Parameters parameters = camera.getNativeCamera().getParameters();
            Camera.Size size = parameters.getPreviewSize();
            YuvImage image = new YuvImage(rxCameraData.cameraData, parameters.getPreviewFormat(), size.width, size.height, null);
            ByteArrayOutputStream baos = new ByteArrayOutputStream();
            image.compressToJpeg(new Rect(0,0,size.width,size.height), 100, baos);
            byte[] jpgData = baos.toByteArray();
            rawBitmap = BitmapFactory.decodeByteArray(jpgData, 0, jpgData.length);
            rawArray.add(rawVideoBitmap);
            rawBitmap.recycle();
        } catch (Exception e) {
                Log.d("EXCEPTION", "ON CAMERA");
        }
    }