ragnraok/RxCamera #20: Problem with result orientation

alyadins, 2016-04-05:
I have problem with front camera on some devices.
On meizu and sony all is ok.
On some chineses devices rotationMatrix is wrong.
[image] Sony
[image] Ulefone Power

alyadins, 2016-04-06:
Initialization
```
	RxCameraConfigChooser chooser = RxCameraConfigChooser.obtain();
		if (getPresenter().isFront()) {
			chooser.useFrontCamera();
		} else {
			chooser.useBackCamera();
		}
		chooser.setAutoFocus(true)
				.setPreferPreviewSize(new Point(640, 480))
				.setHandleSurfaceEvent(true);
		config = chooser.get();
		RxCamera.open(this, config)
				.flatMap(rxCamera -> {
					this.rxCamera = rxCamera;
					return rxCamera.bindTexture(textureView);
				})
				.flatMap(RxCamera::startPreview)
				.observeOn(AndroidSchedulers.mainThread())
				.subscribe(rxCamera -> {
					PhotoActivity.this.rxCamera = rxCamera;
				});
```
requesting
```
	if (rxCamera.isOpenCamera()) {
			rxCamera.request().takePictureRequest(true,
					() -> {
					},
					1080,
					1920,
					ImageFormat.JPEG,
					false)
					.subscribeOn(Schedulers.io())
					.observeOn(Schedulers.io())
					.subscribe(rxCameraData -> {
						if (getPresenter().isFront()) {
							getPresenter().createBitmapFromFront(rxCameraData.cameraData, rxCameraData.rotateMatrix);
						} else {
							getPresenter().createBitmapFromBack(rxCameraData.cameraData, rxCameraData.rotateMatrix);
						}
					});
		}
```

ragnraok, 2016-04-09:
that mens the orientation retrieve from CameraInfo is wrong, can you show me the result of RxCameraConfig.toString?

alyadins, 2016-04-11:
Nexus 5
RxCameraConfig isFaceCamera: true, currentCameraId: 1, preferPreviewSize: Point(640, 480), minPreferPreviewFrameRate: -1, maxPreferPreviewFrameRate: -1, previewFormat: -1, displayOrientation: 90, isAutoFocus: truepreviewBufferSize: -1, isHandleSurfaceEvent: true, cameraOrien: 270, 

Ulefone 
RxCameraConfig isFaceCamera: true, currentCameraId: 1, preferPreviewSize: Point(640, 480), minPreferPreviewFrameRate: -1, maxPreferPreviewFrameRate: -1, previewFormat: -1, displayOrientation: 90, isAutoFocus: truepreviewBufferSize: -1, isHandleSurfaceEvent: true, cameraOrien: 270,

They are the same...

ragnraok, 2016-04-12:
It seem that I need some special fix for this machine, I will find a way to fix this, thanks for your report

gwez, 2016-05-01:
I have similar issue on nexus 6P, if in landscape mode the image is upside down when applying the rotation matrix.

f0restgump92, 2016-09-17:
This issue is happening because of mirroring mode on some devices. Solution for this can be sorted out with https://gist.github.com/9re/1990019, but it depens on device's camera software populating exif information and should work on most of the cases...

ragnraok, 2016-09-18:
@f0restgump92 the exif information only help for saved images, and can't fix the camera preview

somranizaineb, 2017-03-16:
if (camera.isOpenCamera()) {
            camera.request().takePictureRequest(false, new Func() {
                @Override
                public void call() {

                }
            }, 480, 640, ImageFormat.JPEG, false).subscribe(new Action1<RxCameraData>() {
                @Override
                public void call(RxCameraData rxCameraData) {
                    File pictureFileDir = getDir();

                    if (!pictureFileDir.exists() && !pictureFileDir.mkdirs()) {

                        return;
                    }

                    SimpleDateFormat dateFormat = new SimpleDateFormat("yyyymmddhhmmss");
                    String date = dateFormat.format(new Date());
                    String photoFile = "Pic" + date + ".jpg";

                    String filename = pictureFileDir.getPath() + File.separator + photoFile;

                    File pictureFile = new File(filename);
                    Bitmap bitmap = BitmapFactory.decodeByteArray(rxCameraData.cameraData, 0, rxCameraData.cameraData.length);

                    if (camera.getConfig().isFaceCamera) {
                        //to turn face picture taken 
                        Matrix mat = new Matrix();
                        mat.postRotate(-90);
                        mat.preScale(1.0f, -1.0f);

                        bitmap = Bitmap.createBitmap(bitmap, 0, 0, bitmap.getWidth(), bitmap.getHeight(),
                                mat, false);
                    } else {
                        

                        Matrix mat = new Matrix();
                        mat.postRotate(-90);
                        mat.preScale(1.0f, -1.0f);
                        bitmap = Bitmap.createBitmap(bitmap, 0, 0, bitmap.getWidth(), bitmap.getHeight(),
                                rxCameraData.rotateMatrix, false);
                    }

                    try {
                        pictureFile.createNewFile();
                        FileOutputStream fos = new FileOutputStream(pictureFile);
                        bitmap.compress(Bitmap.CompressFormat.JPEG, 100, fos);
                        fos.close();

                        Intent mediaScanIntent = new Intent(Intent.ACTION_MEDIA_SCANNER_SCAN_FILE);
                        Uri contentUri = Uri.fromFile(pictureFile);
                        imageUrls.add(contentUri.toString());
                        mediaScanIntent.setData(contentUri);
                        mActivity.sendBroadcast(mediaScanIntent);

                    } catch (IOException e) {
                        e.printStackTrace();
                    }

                }
            });
        }