ragnraok/RxCamera #24: PeriodicDataRequest updates the available data?

EzequielAdrianM, 2016-04-25:
Hi, after being able to decode the YUV RxCameraData to JPG, I am facing a new issue.

I have set the periodicDataRequest time to 250 milliseconds. But sometimes the CameraData received is the same! So it is not updated?
How can it be posible if the minimum framerate is 15fps?
Any ideas, help?

ragnraok, 2016-04-25:
did you always get the same data from periodicDataRequest?

EzequielAdrianM, 2016-04-25:
No, only if I set the timer to < 250 ms I get the same data but after one or two repetitions it updates

ragnraok, 2016-04-26:
may be your device support preview framerate is too small, you can call Camera.Parameters.getSupportedPreviewFpsRange() to list the all support fps on your device

EzequielAdrianM, 2016-04-26:
I think it is not fps problem because:

    support preview fps range list: (15000,30000) (30000,30000)
    all support preview size: (1920,1080) (1440,1080) (1280,960) (1280,720) (1056,864) (1024,768) (1008,566) (800,600) (800,480) (800,450) (720,480) (640,480) (528,432) (480,320) (480,270) (352,288) (320,240) (176,144)
    getPreviewBufferSizeFromParameter, previewFormat: 17, previewSize: android.hardware.Camera$Size@3be67b8, bitsPerPixels: 12

ragnraok, 2016-04-26:
that mean the frequency of ``onPreviewFrame`` called by the system on your device is too low, so the ``PeriodicDataRequest`` can only return the pervious data for your

closed 2016-04-26 by ragnraok

EzequielAdrianM, 2016-04-26:
Hi, I have checked and onPreviewFrame inside RxCameraPeriodicDataRequest is called at least 27 times per second, so that's not the problem.
I have made a small version of my actual App so you can try it by yourself, for Android Studio...
Try it on your phone and tell me if you are also getting clod images like me! Thank you!
[https://drive.google.com/open?id=0B91djWX7fgiPNmQycE9Ncnhxblk](url)