ragnraok/RxCamera #25: Extension Features

ragnraok, 2016-05-08:
- [ ] Square Preview surface
- [ ] Face Detection
- [ ] Video Record(by MediaRecorder)

EzequielAdrianM, 2016-05-08:
My application uses RxCamera for taking pictures, and I have recently added video recording to my app
That's the code that worked for me and, hope will help this library...

    private RxCamera rxCamera;
    private MediaRecorder mediaRecorder;

    public void startRecording() {
    	if(!prepareMediaRecorder()) {
            Log.d("---", "Failed recording");
            return;
        }
    	try {
    	    mediaRecorder.start();
    	} catch (final Exception e) {
    		Log.i("---", "Exception");
    	}
    }

    public void stopRecording() {
    	mediaRecorder.stop(); // stop the recording
    	releaseMediaRecorder(); // release the MediaRecorder object
    	Log.d("---", "Video Captured!");
    }

    private void releaseMediaRecorder() {
        if(mediaRecorder != null) {
            mediaRecorder.reset(); //clear recorder config
            mediaRecorder.release(); // release the recorder
            mediaRecorder = null;
            rxCamera.getNativeCamera().lock(); //lock the camera for later use
        }
    }

    private boolean prepareMediaRecorder() {
        rxCamera.getNativeCamera().unlock();
        mediaRecorder = new MediaRecorder();
        mediaRecorder.setCamera(rxCamera.getNativeCamera());
        mediaRecorder.setAudioSource(MediaRecorder.AudioSource.CAMCORDER);
        mediaRecorder.setVideoSource(MediaRecorder.VideoSource.CAMERA);
        try {
            mediaRecorder.setProfile(CamcorderProfile.get(CamcorderProfile.QUALITY_1080P));
        } catch (Exception e) {
            try {
                mediaRecorder.setProfile(CamcorderProfile.get(CamcorderProfile.QUALITY_720P));
            } catch (Exception er) {
                mediaRecorder.setProfile(CamcorderProfile.get(CamcorderProfile.QUALITY_480P));
            }
        }
        mediaRecorder.setOutputFile("/sdcard/video.mp4");
        mediaRecorder.setMaxDuration(-1); //Unlimited duration...
        mediaRecorder.setVideoFrameRate(30);
        try {
            mediaRecorder.prepare();
        } catch (IllegalStateException e) {
            releaseMediaRecorder();
            return false;
        } catch (IOException e) {
            releaseMediaRecorder();
            return false;
        }
        return true;
    }

Crysis21, 2016-05-09:
For video recording I integrated MediaRecorder into rxCamera code for a project of mine, a week ago. I will make a pull request in the next days, maybe ragnraok will agree to with the code.