ragnraok/RxCamera #29: Why do you allow backup on AndroidManifest

tupac19861013, 2016-06-07:
Hi Dev,

Can you help me out to set the value of allowBackup variable in application tag inside AndroidManifest file set to false. It is causing issue with my app.

Thank you.

Cheers,
Amila Fonseka

mr-cpp, 2016-06-07:
allowBackup means that your application data will be stored on google data servers
so when user will sign in for example on another device stored data for app will be received of couse if the last is installed
and in other similar cases the same

tupac19861013, 2016-06-07:
Thank you for your promptly reply. So the thing is it should be set from the Application that uses your library, since the company policy is to not to store any data on google servers, I can't build the app when using your library. So isn't it better that option to be removed from the library, so that the app builder has the ability to use it or not to use it.

Cheers,
Amila Fonseka

ragnraok, 2016-06-07:
Ok, I will remove this, the ``allowBackup`` present in the AndroidManifest just because the android studio generated file will contain this by default

monowar1993, 2016-06-07:
Please update the readme.md file. 0.0.3 version do not have face detection. I need it. :)

tupac19861013, 2016-06-08:
Hello Ragnraok, did you commit it, coz I don't see the commit with the flag removed.
Thank you for the effort you put in.

ragnraok, 2016-06-08:
I have committed this, please check out the new version

closed 2016-06-08 by ragnraok

tupac19861013, 2016-06-09:
Hi, I've checked the version is still 0.03, what is the new version....?

Thank you

ragnraok, 2016-06-09:
you should checkout the new code, the version is not update yet

EzequielAdrianM, 2016-06-09:
I've checked out the new code, but I still need you to update the Maven repository in order to definitely implement on Android Studio