ragnraok/RxCamera #30: Multiple Image at a time using rxcamera

AanalShah, 2016-06-07:
Hey can i take multiple image before close the camera at a time ?

how to make brust mode using rxcamera 
pls help me

ragnraok, 2016-06-08:
what do you mean of "brust mode"?

AanalShah, 2016-06-09:
its means u can take a batch of image without closing camera in one time

ragnraok, 2016-06-09:
That is impossible, the system [takePicture](https://developer.android.com/reference/android/hardware/Camera.html#takePicture(android.hardware.Camera.ShutterCallback, android.hardware.Camera.PictureCallback, android.hardware.Camera.PictureCallback, android.hardware.Camera.PictureCallback)) call will definitely stop your preview, or you can use ``oneShotRequest``, but you need to convert the yuv to rgb manually

AanalShah, 2016-06-09:
okk

closed 2016-06-09 by AanalShah