ragnraok/RxCamera #35: samsung devices not focus

spirosoik, 2016-07-04:
I can see that for a plenty of Samsung devices the camera cannot auto-focus for some reason.

ragnraok, 2016-07-04:
could you post your code and the system log?

spirosoik, 2016-07-09:
```java
  /**
   * Inits the camera view
   */
  private void initCamera() {
    //creates the config for the camera
    RxCameraConfig config = RxCameraConfigChooser.obtain().
        useBackCamera().
        setPreferPreviewSize(new Point(640, 480), true).
        setAutoFocus(true).
        setHandleSurfaceEvent(true).
        get();
    // opens the camera
    RxCamera.open(getContext(), config).flatMap(new Func1<RxCamera, Observable<RxCamera>>() {
      @Override public Observable<RxCamera> call(RxCamera rxCamera) {
        return rxCamera.bindTexture(camTextureView);
      }
    }).flatMap(new Func1<RxCamera, Observable<RxCamera>>() {
      @Override public Observable<RxCamera> call(RxCamera rxCamera) {
        return rxCamera.startPreview();
      }
    }).observeOn(new UIThread().getScheduler()).subscribe(new Subscriber<RxCamera>() {
      @Override public void onCompleted() {
      }

      @Override public void onError(Throwable e) {
      }

      @Override public void onNext(RxCamera rxCamera) {
        rxCamera = rxCamera;
      }
    });
  }
```

spirosoik, 2016-07-09:
In general seems that camera doesn't focus in any device and it's a problem

ragnraok, 2016-07-10:
your devices may not support FOCUS_MODE_AUTO, you can try to use ``areaFocusAction`` to set the focus area

spirosoik, 2016-07-11:
@ragnraok users are complaining and the device is Samsung Galaxy S7. As you understand it's very difficult to not support FOCUS_MODE_AUTO such a device.

ragnraok, 2016-07-20:
Hi, have you try the focus mode ``FOCUS_MODE_CONTINUOUS_VIDEO`` or ``FOCUS_MODE_CONTINUOUS_PICTURE``, it may better for the long term services such as barcode scanning or voip

samy-baili, 2016-12-12:
I got the same issue on Nexus 5X, auto-focus doesn't work