ragnraok/RxCamera #38: Could you please release a new version

afaucogney, 2016-07-19:
Latest publish version is 0.0.4 and do not yet support latest feature (Builder)
Latest commit doesn't build on Jitpackio https://jitpack.io/#ragnraok/RxCamera

ragnraok, 2016-07-19:
do you mean the ``RxCameraActionBuilder``?

and I didn't publish this library on jitpack, please use jcenter

afaucogney, 2016-07-19:
RxCameraConfig config = new RxCameraConfig.Builder()
                .useBackCamera()
                .setAutoFocus(true)
                .setPreferPreviewFrameRate(15, 30)
                .setPreferPreviewSize(new Point(640, 480), false)
                .setHandleSurfaceEvent(true)
                .build();

Is not available with the 0.0.4 version

Jitpack just allow to use any version of the lib available on github, even version derived from commit. It is usefull in your case, but as it doesn"t work, I had to put a module with your lib i my app project (it less easy to update)

ragnraok, 2016-07-20:
ok, I will released the new version after some necessary bugs fixed

closed 2016-08-08 by ragnraok