ragnraok/RxCamera #45: Camera Preview Size and taken camera size issue

canberkcakmak, 2016-08-05:
Hi, i'm not using toolbar in my application and, preview streched showing. By the way after taking picture,  photo cropped left and right side. I'm testing 3 device and result is same. here is example pictures : 

[before](http://i.hizliresim.com/EJ7qdB.png)
[after](http://i.hizliresim.com/jnE2OD.jpg)

and  here is my configs : 
```
 RxCameraConfig config = new RxCameraConfig.Builder()
                .useBackCamera()
                .setAutoFocus(true)
                .setPreferPreviewFrameRate(15, 30)
                .setPreferPreviewSize(new Point(640, 480), false)
                .setHandleSurfaceEvent(true)
                .build();
```

How can i solve this?

ragnraok, 2016-08-06:
May be the devices doesn't support 640x480 preview size, can you check the log for tag ``RxCamera.CameraUtil`` ?

canberkcakmak, 2016-08-06:
There are Note 3, s3 , and asus k00l tablet. And demo apk not strec.

ragnraok, 2016-08-06:
So, what about de log? The stretch may happen if the final preview size not match the screen size ratio

closed 2016-08-07 by canberkcakmak