ragnraok/RxCamera #46: How to take a video ?

achavhan, 2016-08-11:


EzequielAdrianM, 2016-08-11:
This is my best-tested implementation of Video Recording with RxCamera.
Checked and working on:
Samsung S4
Motorola G2
Samsung S4 Mini
LG Nexus 5

    private MediaRecorder mediaRecorder;
    private RxCamera camera;
    private Camera.Size sizeCam;
    private RxCameraConfig config;
    private String picName;
    private SimpleDateFormat formatter = new SimpleDateFormat("yyyyMMdd_HHmmss");
    int storagePermissionCheck;

    private void requestRecordVideo() {
        if(!checkCamera()) {
            return;
        }
        if(!prepareMediaRecorder()) {
            Log.d("RXCAM", "FAIL RECORDING");
            return;
        }
        // work on UiThread for better performance
        runOnUiThread(new Runnable() {
            public void run() {
                try {
                    //SEND RECORD SIGNAL!
                    mediaRecorder.start();
                    final int videoMaxTimeMs = 5000;
                    new CountDownTimer(videoMaxTimeMs, 200) {
                        public void onTick(long millisUntilFinished) {}
                        public void onFinish() {
                            mediaRecorder.stop(); // stop the recording
                            releaseMediaRecorder();
                        }
                    }.start();
                } catch (final Exception ex) {
                    Log.i("---","Exception in thread");
                }
            }
        });
    }

    public void openFrontCameraAsync() {
        new Thread(new Runnable() {
            public void run() {
                 config = RxCameraConfigChooser.obtain().
                        useFrontCamera().
                        setAutoFocus(true).
                        setPreferPreviewFrameRate(15, 30).
                        setPreferPreviewSize(new Point(1280, 720)).
                        setHandleSurfaceEvent(true).
                        get();
                RxCamera.open(context, config).flatMap(new Func1<RxCamera, Observable<RxCamera>>() {
                    @Override
                    public Observable<RxCamera> call(RxCamera rxCamera) {
                        camera = rxCamera;
                        return rxCamera.bindTexture(cameraPreview);
                    }
                }).flatMap(new Func1<RxCamera, Observable<RxCamera>>() {
                    @Override
                    public Observable<RxCamera> call(RxCamera rxCamera) {
                        sizeCam = rxCamera.getNativeCamera().getParameters().getPreviewSize();
                        return rxCamera.startPreview();
                    }
                }).observeOn(AndroidSchedulers.mainThread()).subscribe(new Subscriber<RxCamera>() {
                    @Override
                    public void onCompleted() {
                        int newPreviewSize = Math.round((screenWidth * sizeCam.width)/sizeCam.height);
[... 59 lines cut ...]
            }
        }
        mediaRecorder.setOutputFile(picName);
        //Step 5. Set Duration
        mediaRecorder.setMaxDuration(-1);
        try {
            mediaRecorder.prepare();
        } catch (IllegalStateException e) {
            releaseMediaRecorder();
            return false;
        } catch (IOException e) {
            releaseMediaRecorder();
            return false;
        }
        return true;
    }

    private void releaseMediaRecorder() {
        if(mediaRecorder != null) {
            mediaRecorder.reset();
            mediaRecorder.release();
            mediaRecorder = null;
            camera.getNativeCamera().lock();
        }
    }

achavhan, 2016-08-11:
please provide complete code

EzequielAdrianM, 2016-08-12:
That's complete. The only thing you have to do is to paste the code inside your Activity onCreate() and call first openFrontCameraAsync() and add a button to call requestRecordVideo()

closed 2016-08-13 by ragnraok