ragnraok/RxCamera #48: Capture image with Aspect Ratio 4:3

bunhann, 2016-08-18:
Hi, 
I want to take a photo with aspect ratio 4:3. Currently, some device it capture with 4:3 and some with 16:9, so i have problem while scale image which were captured.

Sorry for my bad English. Hope i can get some advice soon.

ragnraok, 2016-08-22:
Can you post a screenshot? And what's the picture size passed to the ``TakePictureRequest`` ?

nonameden, 2016-08-29:
Hi there,

Awesome library, found an issue in the TakePictureRequest class
`private Camera.Size findClosetPictureSize(List<Camera.Size> sizeList, int width, int height) {
        if (sizeList == null || sizeList.size() <= 0) {
            return null;
        }
        int minDiff = Integer.MAX_VALUE;
        Camera.Size bestSize = null;
        for (Camera.Size size : sizeList) {
            int diff = Math.abs(size.width - width) + Math.abs(size.height);
            if (diff < minDiff) {
                minDiff = diff;
                bestSize = size;
            }
        }
        return bestSize;
    }`

height never been used, so I believe it should be:
`int diff = Math.abs(size.width - width) + Math.abs(size.height - height);`

Regards, d

ragnraok, 2016-08-29:
@nonameden you are right, I will fix this

nonameden, 2016-08-29:
Thank you,

Do you have in plans add support for Camera2?

bunhann, 2016-08-30:
Thank you so much.

ragnraok, 2016-08-30:
@nonameden Not yet, but may be I will release a new called RxCamera2 to support the camera2 API

closed 2016-09-04 by ragnraok

uDevel, 2016-12-23:
Is this push to com.ragnarok.rxcamera:lib:0.0.5?
It's in master but not 0.05.