ragnraok/RxCamera #49: Front Camera Touch Focus

ezesolares, 2016-08-29:
Hello, great libraries.

I was wondering if there is a reason why you cant use touch focus while using the front facing camera, it doesnt work on the example app.
Im testing it on my nexus 6p.

I keep getting area focus and metering failed..

Thanks in advance

ragnraok, 2016-08-30:
It is the ``onError`` get called? may be the front camera on nexus 6p does not support area focus and metering, can you post your code here?

ezesolares, 2016-08-30:
Im using the example to test, nothing else. Also, the nexus 6p support touch focus on their own camera app.

The error im getting is 
 D/Example.MainActivity: area focus and metering failed: null

which comes from the areaFocusAction in RxCameraActionBuilder.java
and from the 
if (parameters.getMaxNumFocusAreas() < focusAreaList.size()) {
for which getMaxNum is 0 and focusArea is 1

The Nexus 6p has touch focus on their own camera app and also, i have the same problem with my other test device (Galaxy S5 with Android 5.0).

I dont know where to check if the nexus 6p support area focus and metering using camera 1 api, but it seems that does support it on the camera 2 api.

Thanks in advance

ragnraok, 2016-08-31:
It means that area focus not support for the front camera, at least in camera1 API, for these devices, may be have a better area focus implementation under the camera2 API, so, for the RxCamera, you can't use area focus for these devices' front camera

May be I will develop a new library for the camera2 API...

closed 2016-09-04 by ragnraok