ragnraok/RxCamera #51: Flash doesn't work on lg devices

farhanahmed95, 2016-11-15:
LG MS330 - 5.1.1
LG VS880 v5.1.1
Flash does not fire in app but works properly in stock camera

ragnraok, 2016-11-15:
back or front camera? can you post your code?

farhanahmed95, 2016-11-15:
Camera preview works perfectly but flash never fire when taking picture request for both back and front. this is mostly happening on lg devices also on LG H811.

here is my requestTakePicture function.
```java
private void requestTakePicture() {
            camera.request().takePictureRequest(false, new Func() {
                @Override
                public void call() {
                }
            }, 640,960, ImageFormat.JPEG, true).subscribe(new Action1<RxCameraData>() {
                @Override
                public void call(RxCameraData rxCameraData) {

                    if (!camera.getConfig().isFaceCamera) {
                        rxCameraData.rotateMatrix.setRotate(90);
                    }
                    pictureTakenLayout.setVisibility(View.VISIBLE);
                    cameraPictureImageView.setVisibility(View.VISIBLE);
                    //cameraSwitchBtn.setVisibility(View.GONE);
                    Bitmap bitmap = BitmapFactory.decodeByteArray(rxCameraData.cameraData, 0, rxCameraData.cameraData.length);
                    bitmap = Bitmap.createBitmap(bitmap, 0, 0, bitmap.getWidth(), bitmap.getHeight(),
                            rxCameraData.rotateMatrix, false);

                    cameraPictureImageView.setImageBitmap(bitmap);
                   

                }
            });        
}
```

Thank you for taking out time and looking into this issue. I am waiting for your response.

farhanahmed95, 2016-11-15:
Here is code for my config.
```java
private void startSetup() {
        RxCameraConfig config = new RxCameraConfig
                .Builder()
                .useFrontCamera()
                .setAutoFocus(false)
                .setPreferPreviewFrameRate(15, 30)
                .setPreferPreviewSize(cameraPoints(), false)
                .setHandleSurfaceEvent(true)
                .build();
        RxCamera.open(this, config).flatMap(new Func1<RxCamera, Observable<RxCamera>>() {
            @Override
            public Observable<RxCamera> call(RxCamera rxCamera) {
                camera = rxCamera;
                return rxCamera.bindTexture(cameraView);
            }
        }).flatMap(new Func1<RxCamera, Observable<RxCamera>>() {
            @Override
            public Observable<RxCamera> call(RxCamera rxCamera) {
                return rxCamera.startPreview();
            }
        }).observeOn(AndroidSchedulers.mainThread()).subscribe(new Subscriber<RxCamera>() {
            @Override
            public void onCompleted() {

            }

            @Override
            public void onError(Throwable e) {

                if (e == null)
                    return;
                Logy.e(TAG, e.getMessage());
                Toast.makeText(getContext(), e.getMessage(), Toast.LENGTH_SHORT).show();
                e.printStackTrace();
            }

            @Override
            public void onNext(final RxCamera rxCamera) {
                camera = rxCamera;
            }
        });
        /*Take Picture*/
        cameraTakeBtn.setOnClickListener(new View.OnClickListener() {
            @Override
            public void onClick(View v) {
                requestTakePicture();
            }
        });
        
    }
```

ragnraok, 2016-11-16:
may be your devices doesn't support Camera.Parameters.FLASH_MODE_ON, did you try ``flashAction``?

closed 2016-11-22 by ragnraok