ragnraok/RxCamera #52: Auto White Balance disabled?

EzequielAdrianM, 2016-11-22:
Hi, while opening my RxCamera I found that the imagen was very dark. You can think it's normal, so reopened the app. But the problem persisted. So, without leaving the room and lightning... I opened the Android Camera of my phone and... Surprise! The exposure was perfect! The same happened when I opened Snapchat. The exposure was perfect. My question is: Did you disabled auto exposure? How can this be fixed? Thanks in advance

ragnraok, 2016-11-22:
the white balance is leave un-change in RxCamera, I never disabled this, it depends on your phone

EzequielAdrianM, 2016-11-24:
I noticed that. On android.hardware.camera the WhiteBalance is left to Automatic by default.
The reason of my issue was that Camera API is not "smart enough" to show a better preview.
Solution? Switching to android.hardware.camera2 API where available (API 21+) and the problem was solved. Thanks!

closed 2016-11-24 by EzequielAdrianM

mhassanreza, 2017-08-16:
Can we resolve this issue without using camera2 api?