ragnraok/RxCamera #57: Save image locally

MagdaBadita, 2016-11-30:
Using this method "periodicDataRequest" I tried to save locally the image but the bitmap is null and  I don't understand why?!

 camera.request().periodicDataRequest(1000).subscribe(new Action1<RxCameraData>() {
            @Override
            public void call(RxCameraData rxCameraData) {

                showLog("periodic request, cameraData.length: " + rxCameraData.cameraData.length);
                try {
                    String path = Environment.getExternalStorageDirectory() + "/test" + "_" + (title++) + ".jpg";
                    File file = new File(path);
                    BitmapFactory.Options options = new BitmapFactory.Options();
                    options.outWidth = 1440;
                    options.outHeight = 1920;
                    byte[] theArray = rxCameraData.cameraData;
                    Bitmap bitmap = BitmapFactory.decodeByteArray(theArray, 0, theArray.length, options);
                    //Bitmap bitmap = BitmapFactory.decodeByteArray(rxCameraData.cameraData, 0, rxCameraData.cameraData.length);
                    bitmap = Bitmap.createBitmap(bitmap, 0, 0, 1440, 1920,
                            rxCameraData.rotateMatrix, false);
                    try {
                        file.createNewFile();
                        FileOutputStream fos = new FileOutputStream(file);
                        bitmap.compress(Bitmap.CompressFormat.JPEG, 100, fos);
                        fos.close();
                    } catch (IOException e) {
                        e.printStackTrace();
                    }
                    showLog("PERIODIC - Save file on " + path);
                } catch (Exception e) {
                    e.printStackTrace();
                }
            }
        });

Please could you help me? Thank you!

ragnraok, 2016-12-04:
the ``periodicDataRequest`` only return the raw yuv data, you need to manually convert it to the rgb format

closed 2016-12-04 by ragnraok

MagdaBadita, 2016-12-06:
Thank you for your answer!
I tried to convert it but it seams to not work. The result image is green.

` private void method(File file, byte[] data, int imageWidth, int imageHeight) {
        // the bitmap we want to fill with the image
        Bitmap bitmap = Bitmap.createBitmap(imageWidth, imageHeight, Bitmap.Config.ARGB_8888);
        int numPixels = imageWidth * imageHeight;

// the buffer we fill up which we then fill the bitmap with
        IntBuffer intBuffer = IntBuffer.allocate(imageWidth * imageHeight);
// If you're reusing a buffer, next line imperative to refill from the start,
// if not good practice
        intBuffer.position(0);

// Set the alpha for the image: 0 is transparent, 255 fully opaque
        final byte alpha = (byte) 255;

// Get each pixel, one at a time
        for (int y = 0; y < imageHeight; y++) {
            for (int x = 0; x < imageWidth; x++) {
                // Get the Y value, stored in the first block of data
                // The logical "AND 0xff" is needed to deal with the signed issue
                int Y = data[y * imageWidth + x] & 0xff;

                // Get U and V values, stored after Y values, one per 2x2 block
                // of pixels, interleaved. Prepare them as floats with correct range
                // ready for calculation later.
                int xby2 = x / 2;
                int yby2 = y / 2;

                // make this V for NV12/420SP
                float U = (float) (data[numPixels + 2 * xby2 + yby2 * imageWidth] & 0xff) - 128.0f;

                // make this U for NV12/420SP
                float V = (float) (data[numPixels + 2 * xby2 + 1 + yby2 * imageWidth] & 0xff) - 128.0f;

                // Do the YUV -> RGB conversion
                float Yf = 1.164f * ((float) Y) - 16.0f;
                int R = (int) (Yf + 1.596f * V);
                int G = (int) (Yf - 0.813f * V - 0.391f * U);
                int B = (int) (Yf + 2.018f * U);

                // Clip rgb values to 0-255
                R = R < 0 ? 0 : R > 255 ? 255 : R;
                G = G < 0 ? 0 : G > 255 ? 255 : G;
                B = B < 0 ? 0 : B > 255 ? 255 : B;

                // Put that pixel in the buffer
                intBuffer.put(alpha * 16777216 + R * 65536 + G * 256 + B);
            }
        }

// Get buffer ready to be read
        intBuffer.flip();

// Push the pixel information from the buffer onto the bitmap.
        bitmap.copyPixelsFromBuffer(intBuffer);

        try {
            file.createNewFile();
            FileOutputStream fos = new FileOutputStream(file);
            bitmap.compress(Bitmap.CompressFormat.JPEG, 100, fos);
            fos.close();
        } catch (IOException e) {
            e.printStackTrace();
        }
    }`