ragnraok/RxCamera #62: Preview too dark on dome devices.

mhassanreza, 2017-09-06:
I am using this library on some devices **(nexus 5, galaxy s5 and samsung tablet)** is causing preview too dark and here is the code I was using...
`RxCameraConfig config = new RxCameraConfig.Builder()
            .useBackCamera()
            .setAutoFocus(true)
            .setPreferPreviewFrameRate(15, 30)
            .setPreferPreviewSize(new Point(640, 480), false)
            .setHandleSurfaceEvent(true)
            .build();`
here is what causing preview too dark is this line  **setPreferPreviewFrameRate(15, 30)** I commented this line and preview is fine.
**OR** 
By replacing this code 
`       try {
            List<Integer> frameRates = parameters.getSupportedPreviewFrameRates();
            if (frameRates != null) {
                Integer max = Collections.max(frameRates);
                Toast.makeText(context, "=======>" + max, 1).show();
                parameters.setPreviewFrameRate(max);
            }
        } catch (Exception e) {
            Log.e(TAG, "get parameter failed: " + e.getMessage());
        }
        `
With
`if (cameraConfig.minPreferPreviewFrameRate != -1 && cameraConfig.maxPreferPreviewFrameRate != -1) {
            try {
                int[] range = CameraUtil.findClosestFpsRange(camera, cameraConfig.minPreferPreviewFrameRate, cameraConfig.maxPreferPreviewFrameRate);
                parameters.setPreviewFpsRange(range[0], range[1]);
                parameters.setPreviewFrameRate(15);
            } catch (Exception e) {
                openCameraFailedReason = OpenCameraFailedReason.SET_FPS_FAILED;
                openCameraFailedCause = e;
                Log.e(TAG, "set preview fps range failed: " + e.getMessage());
                return false;
            }
        }`
Inside **RxCameraInternal.java** file. in the following path: "RxCamera-lib\src\main\java\com\ragnarok\rxcamera\RxCameraInternal.java"

**so Question is why setting frame rate not working on some devices?**
>

ragnraok, 2017-09-06:
how did you know the fps setting was not working? can you post some screenshot?

mhassanreza, 2017-09-13:
I am using this library inside my app. and client complaint about this issue so i decided to read android camera api documentation.  I tried many others solutions from stackoverflow and then i get to know that fps was causing this issue.
I am using front camera.
**Device: Samsung Galaxy S5
Model number: SM-G900I
Android version: 6.0.1**

Look at the attached screenshots
[image]
[image]

ragnraok, 2017-09-14:
you can't directly set you preview frame with specific value, you need to call ``getSupportedPreviewSizes`` to query the supported frame rates list and set it accordingly, or you  should set the preview frame rate in the specific range with ``setPreviewFpsRange``, thats how we do in RxCamera, otherwise it may affect the camera metering