ragnraok/RxCamera #65: Getting crash while calling requestTakePicture()

deshmukhrohit11, 2018-04-19:
I am getting following crash while calling requestTakePicture()

04-19 12:07:31.102 28721-28721/com.ragnarok.rxcamera.example E/AndroidRuntime: FATAL EXCEPTION: main
    Process: com.ragnarok.rxcamera.example, PID: 28721
    rx.exceptions.OnErrorNotImplementedException: takePicture failed
        at rx.Observable$27.onError(Observable.java:7923)
        at rx.observers.SafeSubscriber._onError(SafeSubscriber.java:159)
        at rx.observers.SafeSubscriber.onError(SafeSubscriber.java:120)
        at rx.Observable.subscribe(Observable.java:8198)
        at rx.Observable.subscribe(Observable.java:8158)
        at rx.Observable.subscribe(Observable.java:7914)
        at com.ragnarok.rxcamera.example.MainActivity.requestTakePicture(MainActivity.java:239)
        at com.ragnarok.rxcamera.example.MainActivity.access$400(MainActivity.java:57)
        at com.ragnarok.rxcamera.example.MainActivity$5$1.run(MainActivity.java:214)
        at android.app.Activity.runOnUiThread(Activity.java:6161)
        at com.ragnarok.rxcamera.example.MainActivity$5.run(MainActivity.java:210)
        at android.os.Handler.handleCallback(Handler.java:789)
        at android.os.Handler.dispatchMessage(Handler.java:98)
        at android.os.Looper.loop(Looper.java:164)
        at android.app.ActivityThread.main(ActivityThread.java:6592)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:769)
     Caused by: java.lang.RuntimeException: takePicture failed
        at android.hardware.Camera.native_takePicture(Native Method)
        at android.hardware.Camera.takePicture(Camera.java:1488)
        at android.hardware.Camera.takePicture(Camera.java:1433)
        at com.ragnarok.rxcamera.request.TakePictureRequest$1.call(TakePictureRequest.java:85)
        at com.ragnarok.rxcamera.request.TakePictureRequest$1.call(TakePictureRequest.java:49)
        at rx.Observable.subscribe(Observable.java:8191)
        at rx.Observable.subscribe(Observable.java:8158) 
        at rx.Observable.subscribe(Observable.java:7914) 
        at com.ragnarok.rxcamera.example.MainActivity.requestTakePicture(MainActivity.java:239) 
        at com.ragnarok.rxcamera.example.MainActivity.access$400(MainActivity.java:57) 
        at com.ragnarok.rxcamera.example.MainActivity$5$1.run(MainActivity.java:214) 
        at android.app.Activity.runOnUiThread(Activity.java:6161) 
        at com.ragnarok.rxcamera.example.MainActivity$5.run(MainActivity.java:210) 
        at android.os.Handler.handleCallback(Handler.java:789) 
        at android.os.Handler.dispatchMessage(Handler.java:98) 
        at android.os.Looper.loop(Looper.java:164) 
        at android.app.ActivityThread.main(ActivityThread.java:6592) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:240) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:769)

saddahussain, 2019-01-16:
Found any fix?

deshmukhrohit11, 2019-01-16:
Nope.